fdu/STM32F769I-disco_Buildroot #4: No longer builds due to changes in glibc, syncconfig

mcelrath, 2020-03-04:
This no longer builds due to changes in glibc described here:

    https://askubuntu.com/questions/1099392/compilation-of-m4-1-4-10-to-1-4-18-fails-due-to-please-port-gnulib-freadahead-c

as well as the removal of the `syncconfig` make target for the linux kernel. (but you can just smash enter to accept all defaults.

The new LTS of buildroot is 2019.2.9, if simply changing to that succeeds (short of smashing enter) I'll add a PR to this issue.

LubomirBogdanov, 2021-03-23:
Hi, could you please share what you did to make it build with an older buildroot? Simply replacing the version number in the Makefile /and hitting ENTER 1000 times for the config/ fails with "elf2flt.c:414:41: error: macro "bfd_section_vma" requires 2 arguments, but only 1 given" on my machine ...

mcelrath, 2021-03-24:
I really don't know and don't have the hardware handy anymore. I'd be very
surprised if this issue hasn't fixed itself in the meantime.

LubomirBogdanov, 2021-03-30:
OK, so a workaround is to do:
1. Install a 32-bit Ubuntu 14.04 on a virtual machine. Note: Ubuntu 14.04 only, not 14.04.1, nor 14.04.2, nor 14.04.3, etc.
2. git clone ..., make bootstrap
3. My institution has all the ports closed, so 
cd STM32F769I-disco_Buildroot/buildroot
make menuconfig
->Bootloaders-> URL of custom repository -> Change "git://git.denx.de/u-boot.git" to "https://git.denx.de/u-boot.git"
4. Save
5. Then do:
cd ../ 
make build
5. At the end the script will fail with:

cp: cannot create regular file '/srv/tftp/stm32f769//' no such file or directory

BUT the kernel will be built anyways.

Cheers!
L.B.

KiranKanchi, 2022-08-07:
Thank you so much for your insight on this topic LubomirBogdanov. I was able to generate the images and flash the U-Boot on to my EVK. I am not able to load the zImage and dts through TFTP (I am new to tftp and i run into timeout when I try to access the images) did you manage to get the boot completely from SD Card by any chance ?

LubomirBogdanov, 2022-08-08:
Yes, I was able to boot from an SD card only, I haven't tried TFTP. To boot from the SD card you need the zImage and the dts binary on it. If you have formatted the card with FATFS, you should use U-Boot with the following start-up commands:

#define CONFIG_BOOTCOMMAND						\
	"run bootcmd_romfs"

#define CONFIG_EXTRA_ENV_SETTINGS \
	"bootcmd_romfs=setenv bootargs ${bootargs} mmc dev 0 && fatload mmc 0 0xc0700000 /stm32f769-disco.dtb && fatload mmc 0 0xc0008000 /zImage && icache off;" \
	"bootz 0x0c0008000 - 0xc0700000\0"

I have actually done an exercise for my students here:
https://github.com/LubomirBogdanov/PVS/tree/master/05_linux

But the main document "01-STM32F769I_Linux.pdf" is in Bulgarian :-( Hope it can help you somehow...

Regards,
L. B.

KiranKanchi, 2022-08-08:
Thank you so much for your kind help LubomirBogdanov.
Do you by any chance have the contents of the image folder which was configured to boot from for SD card in your repo ?
So hopefully I can just flash the uboot and store zImage and .dtb directly without build just to start with :)

LubomirBogdanov, 2022-08-09:
Yes, I will try to send them to your e-mail, because I cannot attach files here ...

LubomirBogdanov, 2022-08-09:
Btw, if you get into the U-Boot prompt you can try to load the kernel with the following commands:

setenv fdtcontroladdr c0ea0128
printenv
mmc dev 0
fatls mmc 0
fatload mmc 0 0xc0700000 /stm32f769-disco.dtb
fatload mmc 0 0xc0008000 /zImage
icache off
bootz 0x0c0008000 - 0xc0700000

If you have your SD card formatted with ext4 the respective commands will be the same except:

...
ext4ls mmc 0
ext4load mmc 0 0xc0700000 /stm32f769-disco.dtb
ext4load mmc 0 0xc0008000 /zImage
...

KiranKanchi, 2022-08-09:
Hello LB, Thank you so much again for your kind response.
Do you by any chance remember what was the U-boot version that used ?

LubomirBogdanov, 2022-08-09:
Yes, unfortunately at some point the U-Boot stopped working with this board. The last commit of U-Boot that still works is this one:

commit a00d15757d7a513e410f15f2f910cb52333361a3 (HEAD -> master, origin/master, origin/HEAD)
Merge: 810ae23fbc 599f7aa541
Author: Tom Rini <trini@konsulko.com>
Date:   Tue Mar 19 19:58:48 2019 -0400

LubomirBogdanov, 2022-08-09:
And one important note - you have to firstly flash the U-Boot at address 0x08000000, then at address 0x08008000 or it won't work. Also you have to disable SPL support from U-Boot's menuconfig.

LubomirBogdanov, 2022-09-16:
Hey! I managed to build it with Ubuntu 20.04!!! :-)

According to

https://stackoverflow.com/questions/69188643/port-gnulib-fseeko-c-to-your-platform-while-building-the-alexa-auto-sdk-on-ubunt

there is a fix. Here is how I managed to get it to build on a Ubuntu 20.04 ...

git clone https://github.com/fdu/STM32F769I-disco_Buildroot.git
cd STM32F769I-disco_Buildroot
make bootstrap
cd buildroot
make clean
make menuconfig
Select Main menu -> Toolchain -> Binutils Version -> *select*
binutils 2.28.1 -> Save -> OK -> Exit
cd ..
make build

At this point the build of m4 will fail but the m4 sources will be downloaded in output/build/host-m4-1.4.18. So ...
Navigate to the directory STM32F769I-disco_Buildroot/buildroot/output/build/host-m4-1.4.18/ and in "freadahead.c" and "fseeko.c" files change this:

#if defined _IO_ftrylockfile || __GNU_LIBRARY__ == 1 /* GNU libc, BeOS, Haiku, Linux libc5 */

to this:

#if defined _IO_EOF_SEEN || __GNU_LIBRARY__ == 1 /* GNU libc, BeOS, Haiku, Linux libc5 */

In "stdio-impl.h" file add this:

/* Glibc 2.28 made _IO_IN_BACKUP private. For now, work around this
problem by defining it ourselves. FIXME: Do not rely on glibc
internals. */

#if !defined _IO_IN_BACKUP && defined _IO_EOF_SEEN
# define _IO_IN_BACKUP 0x100
#endif

below this:

#if defined __NetBSD__ /* NetBSD */
/* Get __NetBSD_Version__. */
# include <sys/param.h>
#endif

Hasn't been tested on the STM32F769 board yet, but I'm pretty sure it will work.

Regards,
L. B.

KiranKanchi, 2022-09-17:
Hi L.B

Your solution worked! I tried it on my WSL2 OS: Ubuntu 20.04.5 LTS on Windows 10 x86_64.
I would also like to point out that freadahead.c and fseeko.c are found in STM32F769I-disco_Buildroot/buildroot/output/build/host-m4-1.4.18/**lib**

After copying the images output to my sd card, I see the kernel starts up but I am flooded with the below 'can't open /dev/ttyX: No such file or directory' have you experienced anything like this in the past ? any workaround if you know ?

can't run '/etc/init.d/rcS': No such file or directory
can't open /dev/tty2: No such file or directory
can't open /dev/tty3: No such file or directory
can't open /dev/tty4: No such file or directory

Kind regards,
Kiran Kanchi

KiranKanchi, 2022-09-18:
Hi L.B,

Did you ever manage to get the LCD frame buffer to work ?

Kind regards,
Kiran Kanchi

LubomirBogdanov, 2022-09-20:
> Hi L.B,
> 
> Did you ever manage to get the LCD frame buffer to work ?
> 
> Kind regards, Kiran Kanchi

Hi, unfortunately no :-( I didn't even try ...

LubomirBogdanov, 2023-01-30:
Hi again,
just wanted to inform you that today I tried this solution on a STM32F769 disco board and it worked as expected! Cheers!

Ivan-Stefanov-513, 2023-05-29:
Hi,

During my lab exams with @LubomirBogdanov I found how to build this version of Buildroot ( Buildroot 2018.02) under Ubuntu 22.10 and 23.04.

Those are the steps:
1. Download and extract Buildroot source files
2. `cd buildroot`
3. `make clean`
4. `make menuconfig`
5. Select Toolchain > Binutils 2.28.1
6. Deselect Bootloaders > U-Boot
7. Deselect Host utilities > openocd (just install it separately)
8. `cd ..`
9. `make build`
10. Build will stop with following error:
`c-stack.c:55:26: error: missing binary operator before token "("`
`55 | #elif HAVE_LIBSIGSEGV && SIGSTKSZ < 16384`
12. Download m4-1.4.19 from [here](https://ftp.gnu.org/gnu/m4/)
13. Replace files in `/buildroot/output/build/host-m4-1.4.18` with those from downloaded m4-1.4.19 (keep the name of the folder host-m4-1.4.18 just swap the files inside with the new ones from m4-1.4.19)
14. `sudo make build` (sudo is needed because replacing the files in step 13 changes permissions on the files I think so there must be way to fix this so that sudo is not needed)
15. Build will stop again with error about `linux-custom Building`
16. Open `/buildroot/output/build/linux-custom/scripts/dtc`
17. In files `dtc-lexer.l` line 41 and `dtc-lexer.lex.c` line 634 add `extern` before `YYLTYPE yylloc`
18. `sudo make build`
19. Build will stop again
20. Open [this](https://salsa.debian.org/clint/fakeroot/-/blob/24d6b0857396cad87b2cabd32fb8af9ef4799915/libfakeroot.c) and copy lines 97 to 107 and add them to local file `/buildroot/output/build/host-fakeroot-1.20.2/libfakeroot.c`. Those are missing definitins for `_STAT_VER`.
21. `sudo make build`
22. The build must finish successfully with message that it cannot open some TFTP connection (`cp: cannot create regular file '/srv/tftp/stm32f769//': No such file or directory`). You should have `/buildroot/output/images` directory with the generated `rootfs.cpio` and `zImage` files :)

If later you need to rebuild you might get error that `/buildroot/output/target/dev/console` cannot be removed if so use `sudo rm /buildroot/output/target/dev/console` and then `sudo make build`.

Kind regards,
Vankata