fdu/STM32F769I-disco_Buildroot #6: mmc in kernel not support ?

salara1960, 2022-11-28:
Good day.
How can I make support for mmc-card (on sdio2) ?
I edited the file stm32f769-disco.dtb, added the following lines to it:
aliases {
     mmc0 = &sdio2;
     serial0 = &usart1;
};

&sdio2 {
    status="okay";
    cd-gpios = <&gpiod 7 0>;
    cd-inverted;
    pinctrl-names = "default", "opendrain";
    pinctrl-1 = <&sdio_pins_b>;
    pinctrl-1 = <&sdio_pins_od_b>;
    bus-width = <4>;
    max-frequency = <25000000>;
};

and in the file stm32f746.dtb:
sdio_pins_b: sdio_pins_b@0 {
    pins {
        pinmux = <STM32_PINMUX('G', 9, AF10)>, //<STM32F769_PG9_FUNC_SDMMC2_D0>,
                <STM32_PINMUX('G', 10, AF10)>, //<STM32F769_PG10_FUNC_SDMMC2_D1>,
                <STM32_PINMUX('B', 3, AF10)>,//<STM32F769_PB3_FUNC_SDMMC2_D2>,
                <STM32_PINMUX('B', 4, AF10)>,//<STM32F769_PB4_FUNC_SDMMC2_D3>,
                <STM32_PINMUX('D', 6, AF10)>,//<STM32F769_PD6_FUNC_SDMMC2_CLK>,
                <STM32_PINMUX('D', 7, AF10)>;//<STM32F769_PD7_FUNC_SDMMC2_CMD>;
                drive-push-pull;
        slew-rate = <2>;
    };
};

sdio_pins_od_b: sdio_pins_od_b@0 {
    pins1 {
        pinmux = <STM32_PINMUX('G', 9, AF10)>, //<STM32F769_PG9_FUNC_SDMMC2_D0>,
                <STM32_PINMUX('G', 10, AF10)>, //<STM32F769_PG10_FUNC_SDMMC2_D1>,
                <STM32_PINMUX('B', 3, AF10)>, //<STM32F769_PB3_FUNC_SDMMC2_D2>,
                <STM32_PINMUX('B', 4, AF10)>, //<STM32F769_PB4_FUNC_SDMMC2_D3>,
                <STM32_PINMUX('D', 6, AF10)>; //<STM32F769_PD6_FUNC_SDMMC2_CLK>;
                drive-push-pull;
                slew-rate = <2>;
    };
    pins2 {
        pinmux = <STM32_PINMUX('D', 7, AF10)>; //<STM32F769_PD7_FUNC_SDMMC2_CMD>;
                drive-open-drain;
                slew-rate = <2>;
    };
};

sdio2: sdio2@40011c00 {
    compatible = "st,stm32f7-sdio";
    reg=<0x40011c00 0x400>;
    clocks = <&rcc 0 167>;
    interrupts = <103>;
    status="enabled";
    pinctrl-0 = <&sdio_pins_b>;
    pinctrl-1 = <&sdio_pins_od_b>;
    pinctrl-names = "default", "opendrain";
    max-frequency = <24000000>;
};

but the kernel does not "see" this device.
I did something wrong in the description of this device.
What exactly, can you suggest?

pauleffect90, 2023-04-17:
did you manage to get it working?

salara1960, 2023-04-18:
"did you manage to get it working?" - no, I found a project in which mmс support is done correctly -> https://github.com/adrianalin/STM32F769I-disco_Buildroot

salara1960, 2023-04-18:
no, I found a project in which mmс support is done correctly ->
https://github.com/adrianalin/STM32F769I-disco_Buildroot

FlorianHeigl, 2026-05-17:
Hi,

an update for others - more recent versions of buildroot include out of the box support for this board.
See: 

buildroot/board/stmicroelectronics/stm32f769-disco

there's a readme.txt, first use the commands given there to build buildroot for the board, then use flash_sd.sh from the same directory to update the bootloader via st-link to make it boot from SD.

(it'll give an extlinux error with older u-boot so you definitely have to do that)

I tried with buildroot 2024.05 without any modifications.

kernel cmdline is
`[    0.000000] Kernel command line: console=ttySTM0,115200 root=/dev/mmcblk0p2 rw rootfstype=ext2 rootwait earlyprintk consoleblank=0 ignore_loglevel`
kernel is
`Linux buildroot 5.15.108 #1 PREEMPT Sat May 16 20:53:59 UTC 2026 armv7ml GNU/Linux`

It has other quirks
- no eth0 online on boot
- It only creates a tiny 32MB root filesystem so in theory that might even fit in the QSPI flash, but no mtd in /proc or /dev to be seen (I think there's 64MB of that somewhere.